ruiqimao/qmkbuilder #3: Did you forget to check in local.js?

NoahAndrews, 2017-01-11:
Both on Windows and Debian, `npm run build` produces this error: `errorify: Error: Cannot find module 'errorify: Error: Cannot find module './local' from '/home/noah/qmkbuilder/src/const'`. It appears there should be a file called `local.js` in `src/const`, but there is none.

ruiqimao, 2017-01-11:
There should be a `local.json` file in src/const that contains the URLs for the server `/build` route and the path to the folder of presets. I'll update that in the README.

NoahAndrews, 2017-01-11:
Ah, gotcha. Thanks.

ruiqimao, 2017-01-11:
It was left out on purpose because, well, every system has a different configuration. I've updated the README, so you should be able to create that file now.

NoahAndrews, 2017-01-11:
Sounds good, closing.

closed 2017-01-11 by NoahAndrews