ruiqimao/qmkbuilder #21: KC_RSPC is invalid

incogizmo, 2017-05-30:
Im trying to configure Space Cadet shift keys and the left shift can be set, KC_LSPO for open perentheses, but the right shift cannot, KC_RSPC, for close parentheses cannot.

Attempted to export and save the json then reupload but complained the config was invalid.

Interstingly you can set a key as KC_RSPO however this will trigger a compile error when trying to download the hex file.

tcatkins99, 2017-06-21:
I'm new to this code and QMK in general but I think [this line](https://github.com/ruiqimao/qmkbuilder/blob/d6873b732399df35fb8ac8e3c330ae614a9fb681/src/const/keycodes.js#L329) is the issue here. It references KC_RSPO but I don't think that's a valid code. I think it should read:
`'KC_RSPC': new Keycode('KC_RSPC', 'RSPC', []),`

instead of
`'KC_RSPO': new Keycode('KC_RSPO', 'RSPO', []),`

au, 2017-08-29:
Fixed in fe2e9525ad78c255da530921a97cb05544a1ca40, but @incogizmo has yet to close it. Just a ping, maybe you can close this yourself.