ruiqimao/qmkbuilder #23: Ergodox support

indrora, 2017-06-13:
Would love to see Ergodox support. Don't know if it's painful because of the split layout or not.

brandenbyers, 2017-07-12:
You should already be able to do this. Just import the raw data keyboard-layout-editor.com ErgoDox layout to get started.

Import this:

```
[{x:3.5},"#\n3",{x:10.5},"*\n8"],
[{y:-0.875,x:2.5},"@\n2",{x:1},"$\n4",{x:8.5},"&\n7",{x:1},"(\n9"],
[{y:-0.875,x:5.5},"%\n5",{a:7},"",{x:4.5},"",{a:4},"^\n6"],
[{y:-0.875,a:7,w:1.5},"",{a:4},"!\n1",{x:14.5},")\n0",{a:7,w:1.5},""],
[{y:-0.375,x:3.5,a:4},"E",{x:10.5},"I"],
[{y:-0.875,x:2.5},"W",{x:1},"R",{x:8.5},"U",{x:1},"O"],
[{y:-0.875,x:5.5},"T",{a:7,h:1.5},"",{x:4.5,h:1.5},"",{a:4},"Y"],
[{y:-0.875,a:7,w:1.5},"",{a:4},"Q",{x:14.5},"P",{a:7,w:1.5},""],
[{y:-0.375,x:3.5,a:4},"D",{x:10.5},"K"],
[{y:-0.875,x:2.5},"S",{x:1},"F",{x:8.5},"J",{x:1},"L"],
[{y:-0.875,x:5.5},"G",{x:6.5},"H"],
[{y:-0.875,a:7,w:1.5},"",{a:4},"A",{x:14.5},":\n;",{a:7,w:1.5},""],
[{y:-0.625,x:6.5,h:1.5},"",{x:4.5,h:1.5},""],
[{y:-0.75,x:3.5,a:4},"C",{x:10.5},"<\n,"],
[{y:-0.875,x:2.5},"X",{x:1},"V",{x:8.5},"M",{x:1},">\n."],
[{y:-0.875,x:5.5},"B",{x:6.5},"N"],
[{y:-0.875,a:7,w:1.5},"",{a:4},"Z",{x:14.5},"?\n/",{a:7,w:1.5},""],
[{y:-0.375,x:3.5},"",{x:10.5},""],
[{y:-0.875,x:2.5},"",{x:1},"",{x:8.5},"",{x:1},""],
[{y:-0.75,x:0.5},"","",{x:14.5},"",""],
[{r:30,rx:6.5,ry:4.25,y:-1,x:1},"",""],
[{h:2},"",{h:2},"",""],
[{x:2},""],
[{r:-30,rx:13,y:-1,x:-3},"",""],
[{x:-3},"",{h:2},"",{h:2},""],
[{x:-3},""]
```

And you will get this:

[image]

[image]

You may need to adjust the pins or wiring depending on your needs.

indrora, 2017-07-21:
title change;

Split keyboards (Let's Split, Ergodox, Dactyl, etc) are funky. In the case of the Lets Split, the choice was to have two micros (Infinity Ergo style) and with the ErgoDox and Dactyl, an IO expander is used for one or the other hand.

Having generic support for "split hand" keyboards like this might be really helpful.

scycer, 2017-09-27:
Hmm i get the idea but wouldn't it be better that you (and other in your situation) just create 2 different keyboards? one for each hand? as im assuming your flashing each half individually with their own firmware. In the situation of the IO expander i dont think this site could easily support it anyway?

nooges, 2017-09-27:
@scycer It doesn't quite work that way, both halves actually share the same firmware, but it differs from non-split keyboards in that there's additional code for communicating between each other.

scycer, 2017-09-27:
Just looked up the Let's Split again :) makes more sense now. So for example both pro micro's in the Let's Split keyboard actually share the same firmware? And just communicate through the 3.5mm plug? I would wonder how this page could handle the more unique situations like split keyboards as its seems it may be a fair bit of work beyond creating the basic hex file for a single board (i'm building a split keyboard atm but have chosen just to run 12 thin wires from the right to left side, hardwired without a plug)

immeraufdemhund, 2017-10-21:
I too was hoping to do the same thing. I am using the serial connection with 12 rows and 6 columns. I finally was able to get it up and running. I ended up comparing the code files in let_split and going from there. From what I can tell what they do in this case is that they add in the `serial.c` `i2c.c` and a few others in the make file. I person also has to upload the two eeprom images into both pro micros. So instead of just a single hex file a person would get also two eep files. I tell you it was a pain though trying to get the build environment up and running so having this website (which makes it very clear what is going on) would be a HUGE bonus especially now that I have to figure out how I want my different layers. I wish I knew web programming a bit more so I could help out but sadly I don't. There are a few others at my work that are also building their keyboards and might possibly be using the same approach as me.

immeraufdemhund, 2017-10-21:
@scycer I was thinking that in the pins section you could allow for "two" atmega32u's then you could specify with a checkbox or radio group how you'd like to have the two talk (EEHand, Serial, I2C) the pin layout would still only show the single atmega32u's. With this option chosen it would then include the few files from let's split and change the makefile to include them. Then when you build the hex you would need to include the EEP files. As a note I have no clue how lets_split made those. I took them and just put them on.

side note: sorry I didn't say atmega32u(four) I have a short in mykeyboard so I can't actually type that number right now... Fix the short you say???? sure if I hadn't hot glued the case together... 🤕