ruiqimao/qmkbuilder #42: FLIP fails to load hex file - address is out of range

mbbmbbmm, 2018-09-15:
I'm trying to compile a working hex file for the Prime L keyboard using kbfirmware.com. 
I uploaded my config, changed a key, downloaded the compiled hex file, set the Device in FLIP 3.4.7 to ATmega32U4 and tried to load it. - But it generates the error message "address is out of range".  
I then downloaded the default Json-file from 
[https://www.primekb.com/pages/programming](https://www.primekb.com/pages/programming) 
just to make sure but this gives me the same result. It was working fine before - for instance It is possible to load a hex file that I generated last year using the same method. 
Thanks!

PancakeLegend, 2018-09-20:
I'm also having the same 'address is out of range' issue with AVRDUDE /AVRDUDESS with HEX files produced by http://kbfirmware.com/

Apparently the current workaround is to download the ZIP instead of the HEX and compile it yourself after rolling back to AVR-GCC 8.1 from AVR-GCC 8.2.

Disclaimer: I don't understand any of these words, but hope that they may be helpful.

`~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ 

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9587
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file "C:\Users\Andrew\Desktop\firmware.hex"
avrdude.exe: input file C:\Users\Andrew\Desktop\firmware.hex auto detected as Intel Hex
avrdude.exe: ERROR: address 0x800232 out of range at line 1097 of C:\Users\Andrew\Desktop\firmware.hex
avrdude.exe: read from file 'C:\Users\Andrew\Desktop\firmware.hex' failed

avrdude.exe done.  Thank you.
`

ruiqimao, 2018-09-20:
Finally got around to fixing this. Should be fine now!

closed 2018-09-20 by ruiqimao