ruiqimao/qmkbuilder #44: Error on Flashing Pro Micro - address out of range

limingjie, 2018-10-28:
Hi Ruiqimao,

Thank you for your keyboard firmware builder, it is an excellent tool! It greatly helps our newbies on building QMK firmware without the necessity of setting up QMK environment. I am using it for most of new keyboards testing before writing my own QMK code.

**Issue Description**

Recently, a friend of mine found that the firmware generated by the builder does not work on Pro Micro, I did a test and found there is an `address out of range` error.

**Potential Solution**

I've experienced the same issue on my own QMK installation. It took me quite some time to figure out that, **avr-gcc v8** does not work well for Pro Micro, while **avr-gcc v7** works perfectly with the same code. The problem seems related to the bootloader, because **avr-gcc v8** works well on GH60 board.

Could you please help investigate if it is the same issue? We are really appreciate!

**Flashing Log**

```
*** QMK Toolbox (http://qmk.fm/toolbox)
    Supporting following bootloaders:
     - DFU (Atmel, LUFA) via dfu-programmer (http://dfu-programmer.github.io/)
     - Caterina (Arduino, Pro Micro) via avrdude (http://nongnu.org/avrdude/)
     - Halfkay (Teensy, Ergodox EZ) via teensy_loader_cli (https://pjrc.com/teensy/loader_cli.html)
     - STM32 (ARM) via dfu-util (http://dfu-util.sourceforge.net/)
     - Kiibohd (ARM) via dfu-util (http://dfu-util.sourceforge.net/)
*** Caterina device connected
*** Attempting to flash, please don't remove device
    Found port: /dev/cu.usbmodem145410
>>> avrdude -p atmega32u4 -c avr109 -U flash:w:/Users/mingjie/Downloads/gh60.hex:i -P /dev/cu.usbmodem145410 -C avrdude.conf
    avrdude: warning at avrdude.conf:14976: part atmega32u4 overwrites previous definition avrdude.conf:11487.
    
    Connecting to programmer: .
    Found programmer: Id = "CATERIN"; type = S
        Software Version = 1.0; No Hardware Version given.
    Programmer supports auto addr increment.
    Programmer supports buffered memory access with buffersize=128 bytes.
    
    Programmer supports the following devices:
        Device code: 0x44
    
    avrdude: AVR device initialized and ready to accept instructions
    
    Reading | ################################################## | 100% 0.00s
    
    avrdude: Device signature = 0x1e9587
    avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
    avrdude: erasing chip
    avrdude: reading input file "/Users/mingjie/Downloads/gh60.hex"
    avrdude: ERROR: address 0x80024b out of range at line 1149 of /Users/mingjie/Downloads/gh60.hex
    avrdude: read from file '/Users/mingjie/Downloads/gh60.hex' failed
    
    avrdude done.  Thank you.
```

Thank you!
Mingjie

PancakeLegend, 2018-10-31:
This appears to be a return of [this issue](https://github.com/ruiqimao/qmkbuilder/issues/42), closed on Sept 20. Am experiencing the same thing using AVRDudess. Confirmed that old hex files still flash, new hex files produce 'out of range' error.

Robbi-Blechdose, 2018-11-02:
Can confirm, .hex file is also 48KB so it definetely will not fit.

ruiqimao, 2018-11-09:
Reverted to avr-gcc 7.3. Will keep an eye out for 8.3.

closed 2018-11-09 by ruiqimao

limingjie, 2018-11-17:
Thank you so much @ruiqimao ! Confirmed the firmware is working on Pro Micro.