ruiqimao/qmkbuilder #60: Still alive?

iandoug, 2020-08-09:
Hi

Is this project still alive? The link to http://www.ruiqimao.com/ gives 404.

The compile page does not download hex or zip ... don't know if this is because of my non-standard layout or if something on the site is broken ...

Can't use the qmk_configurator at the moment because it doesn't know my physical layout.

Thanks, Ian

ruiqimao, 2020-08-09:
Nothing should have changed, and the site is still working for me. Can you confirm that you're still having this issue?

iandoug, 2020-08-09:
There is a message about "Not every key has a unique position in the wiring matrix."  ... but it doesn't say what is missing.

Perhaps that is why the download hex or zip file buttons don't work?

I did save the configuration file.

[x64h-3.zip](https://github.com/ruiqimao/qmkbuilder/files/5048198/x64h-3.zip)

Thanks, Ian

iandoug, 2020-08-11:
Made a different layout, was able to download. Thanks a mill.

Now to study the file and try and figure out things about QMK :-)

closed 2020-08-11 by iandoug