ruiqimao/qmkbuilder #69: Error: Cannot find module

gitsang, 2021-03-14:
```
# npm install
npm WARN react-codemirror@0.3.0 requires a peer of react@>=0.14 <16 but none is installed. You must install peer dependencies yourself.
npm WARN react-codemirror@0.3.0 requires a peer of react-dom@>=0.14 <16 but none is installed. You must install peer dependencies yourself.

audited 475 packages in 3.024s

9 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities

# npm run deploy

> qmkbuilder@1.0.0 deploy /home/sang/operation/qmkbuilder
> cross-env NODE_PATH=./src NODE_ENV=production node deploy.js

errorify: Error: Cannot find module './local' from '/home/sang/operation/qmkbuilder/src/const'
```

gitsang, 2021-03-14:
How can I fix it

keebzdotnet, 2021-03-24:
I have the same error and can't get it running

closed 2021-05-13 by gitsang