ruiqimao/qmkbuilder #70: Thank you

AarynSmith, 2021-03-14:
Thank you for running this site for so many years. I just got into building my own keyboard and was using kbfirmware to build, and help me learn to use the source. Thanks for all the hard work you did, and good luck with your future project.

andywarburton, 2021-03-15:
Same here! This tool was essential to me getting started with building my own firmware - there isn't another tool out there! How can we help you get up and running again?

ruiqimao, 2021-03-15:
Thanks for the support! I've actually brought the website back online,
since there are actually more users than my analytics were reporting.

If you would like to host your own copy, the README now links a Docker
image that you can use to run the last live version.

github1man, 2021-03-16:
You're back! I`m so happy! Don't leave us again!

nooges, 2021-03-16:
@ruiqimao Can you specify a license for your code as requested in https://github.com/ruiqimao/qmkbuilder/issues/33? Thanks!

closed 2021-03-26 by AarynSmith